ixmilia/dxf #157: Example on how to use DxfImage

ponahoum, 2021-04-07:
Hi,

I am trying to use the DxfImage entity to insert an image dependency into my DXF but I cannot make it work. The DXF output seems not to contain the image attributes that I put. Does anyone have a working example?

```
DxfImage image = new DxfImage(imagePath: "test.png", new DxfPoint(0,0,0), 256,256, new DxfVector(250,250,0));
dxfFile.Entities.Add(image);
```

Thanks !

ponahoum, 2021-04-07:
Ok I actually found out reading the tests that to make it work you need to add `file.Header.Version = DxfAcadVersion.R14` :

```
            var file = new DxfFile();
            file.Header.Version = DxfAcadVersion.R14; // DxfImage is only supported on >= R14
```

ponahoum, 2021-04-07:
Although the attributes are now in the DXF, i can't manager to make autocad open the DXF with the image as an external reference. Any idea on how to achieve this ?

brettfo, 2021-04-07:
`IMAGE` entities are problematic because AutoCAD is _really_ picky about what the file looks like, and I haven't yet been able to determine exactly what that is.  Using your example code an image displays correctly both in the ODA Viewer app as well as LibreCAD, but not AutoCAD 2016 (I don't have any other programs to test with.)  I even did a Save As from LibreCAD and AutoCAD _still_ couldn't open it; it seems that `IMAGE` is a problem for everybody.

ponahoum, 2021-04-08:
I tried with AutoCAD 2020 and it ignores the reference as well. Although I tried with AutoCAD 2020 and netDxf  / ezDxf and it works well.