ixmilia/dxf #168: How to save dxf with "GB2312"code?

tenglong11, 2021-11-22:
I have create a dxf  with layer ="长江"(chinese character), and save it. When I open it with Notepad++,I found it's code is "UTF-8",but the layer name code is "Unicode" and present "\U+957F\U+6C5F" which is the Unicode of "长江". How  can  I save all dxf with "GB2312" or save all dxf with "UTF-8" code(contain the layer name)?

brettfo, 2021-11-22:
According to [the spec](https://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-2553CF98-44F6-4828-82DD-FE3BC7448113) under "Storage of String Values", DXF files can only be saved as ASCII or CIF (which is the `\U+...` values you're seeing) _unless_ the file is saved as version R2007 or higher, in which case UTF-8 is used.  So if you need UTF-8 and don't care too much about the version, you can do this:

``` csharp
DxfFile file = new DxfFile();
file.Header.Version = DxfAcadVersion.R2007;
// ...
file.Save(...);
```

tenglong11, 2021-11-23:
@brettfo  Thank  you for your help，I have solve the confusion under your suggestion。

closed 2021-11-23 by tenglong11