ixmilia/dxf #169: Exception with some dxf file

mattia589, 2021-12-12:
Hi, i tried to load some dxf file using this library but after the loading phase an exception occurs.
Here the exception message for both the dxf files "Unsupported value '1.000000000000000E 20' for code '10'".
[Bug.zip](https://github.com/ixmilia/dxf/files/7699089/Bug.zip)

brettfo, 2021-12-12:
Thank you for reporting the bug.

The value listed isn't a format that my library expects, but I can certainly make it work.

Where did you get the files, or what software produced them?  I'd like to do some more exploratory testing with it to see if there are other differences that I need to address.

mattia589, 2021-12-12:
I downloaded it from a free dxf download website, I do not remember the website name.
I take the opportunity to ask if you have some testing dxf file to share with me, them are helpful to improve the code I wrote. 

Thanks in advance

brettfo, 2021-12-12:
I've fixed the bug in commit aebcf144aad093336aadb421659c189a3d31acde and pushed the changes to `main`.

I don't have much by way of examples, but the README to this repo has a small sample and you can also look at the `IxMilia.Dxf.Tests` project for ideas.

closed 2021-12-12 by brettfo