ixmilia/dxf #171: Unexpected extra values for code

kub1x, 2022-01-13:
Hello when trying to parse enclosed dxf file below, from this class: `DxfMentalRayRenderSettingsGenerated.cs`, I'm getting following `Debug.Assert` failures AND the result is incorrectly positioned as shown in the picture. I don't know, whether the two symptoms are related, but they occur only together. Even if this is just the missing codes (unrelated to the visual error, which might be in our SW) it might be of some use for this library. The DXF image shows correctly in CorelDraw. 

```
Unexpected extra values for code 290:  12
Unexpected extra values for code 1:     3
Unexpected extra values for code 290:  12
Unexpected extra values for code 40:   11
```

[image]

[filtr_cloud.dxf.txt](https://github.com/ixmilia/dxf/files/7863473/filtr_cloud.dxf.txt)

Thanks, Jakub

brettfo, 2022-01-15:
## Unexpected extra values

According to [the spec](https://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-69B42C9D-062D-4FEC-952E-CA0DDF7B4831) the `MENTALRAYRENDERSETTINGS` object was deprecated in R2016 and it does appear to have duplicate values in this file.

Starting at line 27807 in your file is where the bulk of that object begins, and lines 27811-27817 are 4 values in a row with code 290, but the spec says it expects (290, 90, 290, 290), so there's either a typo in the spec (it's happened before) or the software used to generate this file is using the wrong code (290 instead of 90).

Then lines 27819-27822 are 2 values with code 1, but again the spec only expects a single value there.

The rest of the object has similar issues.

Long story short, the file doesn't match the spec and I don't know who's right and who's not, but none of those differences appear to be causing the issues with the circles.

## Circles appearing below the line

The circles appear to come from a block reference and I'm not sure why they're not aligned, I'll have to look into it some more.

kub1x, 2022-01-16:
Thank you for the prompt answer. I wasn't sure if this bugreport would help in any way and it feels the deprecated stuff is unrelated and does not really matter. 

> The circles appear to come from a block reference and I'm not sure why they're not aligned, I'll have to look into it some more.

This would be error on our side after all. We dug deeper and it turned out we are not handling the `DxfImport`s correctly since we were not aware, they can also contain transformations of the imported object which is what caused the misspositioning. 

Again, thanks for your help and sorry it wasn't much of a use here.

closed 2022-01-16 by kub1x

brettfo, 2022-01-17:
> we were not aware then can also contain transformations

There's obviously something I could do better here to make this discoverable.  `DxfInsert` has a property `Entities` that returns the raw child entities, and I assume this is what you were using along with the `Location` property, but perhaps it would be beneficial to also add something like `AdjustedEntities`, `TransformedEntities`, etc. that returns a copy of the child entities with the appropriate transformations applied.  Since you've recently used this feature, what's something that would make sense to you and your scenario?  I'd like to make this easier for everybody going forwards.

kub1x, 2022-01-17:
Well, it depends how thin the library is supposed to be. If I would be interested in all-and-every DXF feature parsed as is, current state - the thin wrapper - would fit my needs as it simply parses the DXF as is and it does it really well. 

Now our intention is to parse the DXF and convert it to our inner representation. We want to get Lines, Arcs, Circles, Splines, Polylines and we also neeed Names of layers they are assigned to. With Arcs (separate entities or within polylines) we want start point, end point, center, radius and sweep direction. We implemented some algorithm to split splines into polylines. We replace single Lines, Arcs and Circles with Polylines too. Eventually we work with the final list of polylines, like it was CNC G-code program and feed it to our laser machine. But that is work on our side we counted on. 

We surely are worried to miss some DXF feature like we did now. We do not really care for `DxfInsert`s and its transformations, `LWPolyline`s or various other types of the basic entities, rendering options ViewPorts or other stuff. 

If the `DxfFile` had a method `GetPureEntities()` that would return similar list like `Entities` only "flatten" with all transformations and substitutions applied, with only one kind of polylines, etc. , I would definitely use it. 

Adding the `TransformedEntities` to `DxfInsert` would  really help with this current issue, but I would still be alerted for other DXF features I might miss. Users are very creative ;-) But perhaps there isn't much more DXF features like this.

brettfo, 2022-01-17:
Thank you for the detailed reply, I'll consider each of your points; I particularly like the idea of `GetPureEntities()` or similar to return the contents of the DXF in a simplified/reduced manner, possibly with various options, e.g., keep polylines or explode into individual segments, reduce curves to N straight segments, etc.

In the meantime if there's a feature that you really want before I've gotten to it, and you have the time to implement (and if your agreement with your employer/client allows it), I'm always open for PRs.