ixmilia/dxf #172: How Convert Bezier curve to DxfSpline

mattia589, 2022-01-13:
In my code i managing the cubicBezier with 4 points (startPoint, control poin1, control point 2, end point) and quadraticBezier 3 points (startPoint, control poin1, end point). I need to convert them to DxfSplines. Is there a easy way to do it? Do you have suggestions about it? I applied to the DxfSpline control points the bezier control points but the dxf file created with this entity does not show me the spline.

brettfo, 2022-01-15:
Much of the math of Bezier curves and Splines is beyond my ability, but I might be able to point you in the right direction.

## Cubic Beziers

In another related repo of mine [ixmilia/bcad](https://github.com/ixmilia/bcad) I have the concept of Beziers/Splines and for cubic Beziers (also called curves of degree 4) I've found I've had some success in combining curves of the same degree; that code is [here](https://github.com/ixmilia/bcad/blob/4b03a07f8e17fbd26fe0621a13cb9d5a8985a2c7/src/IxMilia.BCad.Core/Entities/Spline.cs#L119-L154) if you want to look at it, but I think by far the easiest thing to do would be to create one `DxfSpline` for each cubic Bezier and I _think_ you might get something like this to work:

``` csharp
dxfSpline.KnotValues.Add(0.0); // note 4 copies of 0.0 because the "degree" of the curve is 4
dxfSpline.KnotValues.Add(0.0);
dxfSpline.KnotValues.Add(0.0);
dxfSpline.KnotValues.Add(0.0);
dxfSpline.KnotValues.Add(1.0);  // and here are 4 copies of 1.0 for the same reason
dxfSpline.KnotValues.Add(1.0);
dxfSpline.KnotValues.Add(1.0);
dxfSpline.KnotValues.Add(1.0);

// now add the 4 control points
dxfSpline.ControlPoints.Add(myCubicBezier.StartPoint);
dxfSpline.ControlPoints.Add(myCubicBezier.ControlPoint1);
dxfSpline.ControlPoints.Add(myCubicBezier.ControlPoint2);
dxfSpline.ControlPoints.Add(myCubicBezier.EndPoint);

// and ensure the degree is set
dxfSpline.DegreeOfCurve = 4;
```

This leaves the open question of what to do with the `FitPoints` collection on the `DxfSpline` entity, and frankly, I have no idea; you'll have to experiment a bit to see what happens.

## Quadratic Beziers

I don't really know about quadratic Bezier curves, but I can fathom a guess.  Given what I did with cubic/degree 4 curves you might have some luck doing something similar with a quadratic curve:

``` csharp
dxfSpline.KnotValues.Add(0.0); // note 3 copies of 0.0 because the "degree" of the curve is 3
dxfSpline.KnotValues.Add(0.0);
dxfSpline.KnotValues.Add(0.0);
dxfSpline.KnotValues.Add(1.0);  // and here are 3 copies of 1.0 for the same reason
dxfSpline.KnotValues.Add(1.0);
dxfSpline.KnotValues.Add(1.0);

// now add the 3 control points
dxfSpline.ControlPoints.Add(myQuatraticBezier.StartPoint);
dxfSpline.ControlPoints.Add(myQuatraticBezier.ControlPoint);
dxfSpline.ControlPoints.Add(myQuatraticBezier.EndPoint);

// and finally the degree
dxfSpline.DegreeOfCurve = 3;
```

The same caveat about the fit points applies; I don't know how they're expected to be used.

If the above doesn't work there is a really good primer on Bezier curves [here](https://pomax.github.io/bezierinfo/), and there's even a section titled [Lowering and elevating curve order](https://pomax.github.io/bezierinfo/#reordering) which I'd imagine you could use to "lift" the quadratic, i.e., degree 3 curve to a cubic, i.e., degree 4, and then write as above.

## Conclusing

tl;dr - I don't really know, but I can guess.  If you get something to work I'd love to hear what you did and even add an example to this repo to help other people.

mattia589, 2022-01-19:
Thanks a lot for your answer.
I have tested it and seems work fine for my topic. I would like to use this opportunity to ask you also the conversion from DxfSpline to Bezier curves. How did you reach it?

brettfo, 2022-01-20:
Using the Bezier curve primer I mentioned above I came up with this which I use in another repo of mine:

Given curve degree, control points, and knot values I create a spline [here](https://github.com/ixmilia/bcad/blob/dfc6a0ce7546062c81b063336e6fc5529229f162/src/IxMilia.BCad.Core/Entities/Spline.cs#L22).

Near the end of that function the spline is turned into a collection of Bezier curves [here](https://github.com/ixmilia/bcad/blob/dfc6a0ce7546062c81b063336e6fc5529229f162/src/IxMilia.BCad.Core/Entities/SplineBuilder.cs#L98).

Note that I only support curves of degree 3 (i.e., 4 control points) in that code so you may have to adapt it to your degree 2 (3 point) curves.

closed 2022-03-14 by mattia589