ixmilia/dxf #174: How can you add grade reference points (e.g. "# 7") to the DXF?

elimtailor, 2022-02-08:
We want to grade DXFs programmatically, but the DXF files don't have grade reference points to start (we have to choose them). Is there a way we can add those to the DXF object and save it back to file with a subset of points denoted as grade reference points?

Thanks!

brettfo, 2022-02-08:
I'm not sure what you mean by grading a reference point as number 7, etc.  Can you provide sample files, ideally a before and after of the same file with graded reference points so I can better understand?  If you can't share the files publicly here, you can email them directly to me at `brett.forsgren@outlook.com` and I'll keep them private and delete them as soon as I'm done.

elimtailor, 2022-02-25:
Thanks @brettfo for resolving my issue over email! I wanted to follow up for anyone else having the same issue:

```
var myFile = DxfFile.Load(“Example2b.dxf”);

foreach (var myBlock in myFile.Blocks)

{

    for (int i = 0; i < myBlock.Entities.Count; i++)

    {

        var myEntity = myBlock.Entities[i];

        if (myEntity is DxfModelPoint myPoint)

        {

            var pointReferenceText = GetPointReference(myPoint);

            if (pointReferenceText != null)

            {

                // create the TEXT item at the same location as the point with the correct text…

                var referenceText = new DxfText(myPoint.Location, pointReferenceText);

                // …and insert it right after in the list

                myBlock.Entities.Insert(i + 1, referenceText);

            }

        }

    }

}

myFile.Save(“Example2b_graded.dxf”);

```

closed 2022-02-25 by elimtailor