ixmilia/dxf #176: LWPolyline is loaded as separate entities

kelyamany, 2022-03-15:
I'm trying to load a dxf file with a dxf version AC1009 with supposedly 2 entites only of LWPolyline type. The problem is that I'm getting the LWPolyline in separated entities. 

For example, this snippet will return me 31 entities of LWPolyline when the file is supposed to only contain one that match the query
```csharp
IEnumerable<DxfEntity> inside = dxf.Entities.Where(e =>e.Layer.ToUpperInvariant() == "INSIDE");
```
The sample dxf I'm using for testing:
[sample_dxf.zip](https://github.com/ixmilia/dxf/files/8254694/sample_dxf.zip)

Is there is a way to merge these entites without having to do that manually?

brettfo, 2022-03-15:
Opening your attached file only shows 2 entities, one on layer `OUTSIDE` and the other on `INSIDE`.  Are you using the latest version of this library?

kelyamany, 2022-03-16:
> Opening your attached file only shows 2 entities, one on layer `OUTSIDE` and the other on `INSIDE`. Are you using the latest version of this library?

I'm using version 0.8.2, I think it's the latest one
The problem is that the single element in the `INSIDE` layer is loaded as 31 separate polylines

brettfo, 2022-03-16:
Can you show me a screenshot of the debugger?  The latest on `main` only loads 1 polyline entity on the "INSIDE" layer which then has 10 vertices, so I have no idea where 31 of anything could come from.

kelyamany, 2022-03-18:
You are right I couldn't replicate the issue with the attached file, it seems that I missed something else in the original file
Sorry for that

closed 2022-03-18 by kelyamany