ixmilia/dxf #179: MText: Missing DXF group code: 90

rocchellid, 2022-05-06:
In doubt that the problem was my management, I do nothing only open a DXF and rewrite the MText and the problem is this that corrupts the reading of the file. There is no problem with Text.
I tried to save with some CAD versions but nothing changes.
I also called Header.SetDefaults () and ViewPorts.Clear () with no results.
I only found a reference to an identical problem with another package but with no details on how it was fixed (https://forum.aspose.com/t/multiline-text-in-dxf-saved-by-aspose-cad-is-corrupt/213358)
Regards,
Davide

brettfo, 2022-05-06:
Can you attach two files, both before and after the MTEXT corruption?  I might be able to look at the differences and figure out what's going on.  If you don't want the files to be public you can email them directly to me at brett.forsgren @ outlook.com and I'll keep them private and delete them as soon as I'm done.

rocchellid, 2022-05-06:
I did another 2 tests even simpler:

1st:

> var a = DxfFile.Load(File);
> a.Entities.Where(x => x.EntityType == DxfEntityType.MText).ToList().ForEach(x =>
> {
>  if (a.Entities.Remove(x))
>  {
>   a.Entities.Add(x);
>  }
> });
> a.Save(Path.Combine(Path.GetDirectoryName(File), "aaa.dxf"));

result: **file corrupted**

2nd even more basic:

> var a1 = DxfFile.Load(File);
> a1.Save(Path.Combine(Path.GetDirectoryName(File), "bbb.dxf"));

result: **file corrupted**

AutoCAD error detail:

> The following error was encountered while reading
> in MTEXT starting at line 2414:
> Missing DXF group code: 90
> Invalid or incomplete DXF input -- drawing discarded

I created file with AutoCAD 2015 and put only a multiline text.

Regards,
Davide

[Test.zip](https://github.com/ixmilia/dxf/files/8642235/Test.zip)

brettfo, 2022-05-06:
Thank you for the file!  Unfortunately it seems that AutoCAD isn't following [their own spec](http://images.autodesk.com/adsk/files/autocad_2013_pdf_dxf_reference_enu.pdf) (page 139), so this may take me a while to figure out.

If you're interested, the MText in the working file (Test MText.dxf) is on lines 2189 to 2258, and in the non-working file (bbb.dxf) is on lines 2413 to 2524, but the one that this library writes _does_ contain a code 90 on line 2465.

rocchellid, 2022-05-06:
I've done some tests.
The most trivial was to remove the 90 (with its value) but nothing has changed.
I then exported with another package and it works; so I took the MTEXT part of the save that goes and put in place of your MTEXT part and now yours goes too; so the problem is (should) only in the MTEXT piece.
I manually added 90 with the values 0,1,2 and it keeps going (I have attached the files).
I also made a comparison of your save and I attached the report.
Unfortunately I was not able to do an identical export (in the one done with the other package the coordinates are all rounded) and the layer is different but otherwise nothing should change.
Hope this can help you find the problem faster.
One last note: AutoCAD Viewer 2020 does not open them, but AutoCAD 2015 does.
Regards,
Davide

[Test MText alternative +90.zip](https://github.com/ixmilia/dxf/files/8643337/Test.MText.alternative.%2B90.zip)

[Report.zip](https://github.com/ixmilia/dxf/files/8643342/Report.zip)

brettfo, 2022-05-10:
Thank you for figuring this out!  I just pushed commit fef38922feb31caa9407a0357016f30a127f9c3c to `main` that fixes this.