ixmilia/dxf #185: Code 46 not readed from MTEXT

rocchellid, 2022-06-23:
Code 46 (Defined annotation height) is very important but is not read directly (instead I would say it should be); is put in the ExcessCodePairs which is protected therefore "unreadable".
Regards,
Davide

rocchellid, 2022-06-27:
Hi, I did the file with AutoCAD 2015. I also did a test with AutoCAD LT 2020 and same thing. The official documentation is probably out of date.
**Code 46 is always present**: by default it is 0.0 which means that the height automatically adjusts it; if you set it (you cannot manually but only graphically by moving the delimitation triangles) then it writes the relative data.
I have attached test files.
Regards,
Davide

[Test.zip](https://github.com/ixmilia/dxf/files/8989999/Test.zip)

brettfo, 2022-06-29:
Thank you for this.

Since I've never used that feature (or even knew that code 46 existed), can you give me some ideas on what to call that property that would be meaningful to developers and hint at its use?  Some quick ones I just thought of:

- `ManualVerticalAdjustmentHeight`
- `ManualHeight`
- `VerticalHeightOverride`

In selecting a name I'd like to capture the concept of it being an override of code 43 (which I named `VerticalHeight`).

rocchellid, 2022-06-30:
In autocad it is called DefinedHeight.
43 is always sigle line font height, also if you have many lines.
Regards,
Davide

brettfo, 2022-06-30:
I just added commit e2b079f857559e0c352556dc7093d4d008a9d851 to `main` that adds the property `DefinedHeight` with code 46.  Can you give this a shot?

rocchellid, 2022-07-01:
It does not work: in the file is 6000 but the variable is 0.
Regards,
Davide

[DE.zip](https://github.com/ixmilia/dxf/files/9027118/DE.zip)

brettfo, 2022-07-01:
Good catch, I forgot I had implemented a custom MTEXT reader.  Fixed in commit 514aac62f99c5a01690d048701de9813adf67769.

closed 2022-07-01 by brettfo