ixmilia/dxf #186: More indexes in code 1, 40 and 290

rocchellid, 2022-07-18:
I have a dxf in which I have::

- 1 index  code 1 more
- 1 index code 40 more
- 2 indexes code 290 more

I am sending you the DXF file in private because I cannot publish it here.
Regards,
Davide

[image]
[image]
[image]

rocchellid, 2022-07-18:
The problem isn't in 0.8.3; I found it with the latest sources updates I downloaded ...

brettfo, 2022-12-03:
## Short Version

I've found the issue and will start working on it now.

## Extra Code 290

Thank you for pointing this out.  Turns out the spec has different layouts under subclass marker `AcDbRenderSettings` as specified in code 90.  The spec for [`MENTALRAYRENDERSETTINGS`](https://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-69B42C9D-062D-4FEC-952E-CA0DDF7B4831) (which is the object type specified in your sample file) explicitly lists class version 1 (row 7 in that table), but the file you sent me specifies class version 2 (line 347630, if you open it in a text editor), and the order of the codes just happens to match the `AcDbRenderSettings` listed in object [`RAPIDRTRENDERSETTINGS`](https://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-427599C5-9562-4384-82E6-8B7B87492C57).  I'll work on fixing the reader for `AcDbRenderSettings` to handle the different class version numbers.

## Extra Code 1

This problem appears to come from the code 290 issue above; version 2 of `AcDbRenderSettings` has an extra code 1 value which messes up the subsequent count.  This should be fixed when I fix the code 290/class version 2 problem.

## Extra Code 40

The object in the file you sent me has 2 code pairs at the very end that the spec doesn't account for: 290, and 40.  I'll adjust the reader to allow these unknown fields to exist so the file can be read.

brettfo, 2022-12-03:
Fixed in `main`.

closed 2022-12-03 by brettfo