ixmilia/dxf #190: spline closed attribute error

Illicitl, 2022-12-03:
When I read the dxf, dxfspline's closed property is wrong, autocad view is closed, but read false, also, the location of the dxftext is wrong.

[error.zip](https://github.com/ixmilia/dxf/files/10145790/error.zip)

brettfo, 2022-12-03:
## Spline Closed Property

Sure enough debugging the file you attached shows all splines as having the `IsClosed` flag set to `false`, but looking at the DXF file in a text editor (lines 2221-2222) shows that code `70` is `0` and according to [the spec](https://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-E1F884F8-AA90-4864-A215-3182D47A9C74) this is the correct interpretation of that flags field.

[image]

The file has 809 `SPLINE` entities, and of those, the first and last control points of 775 of them have the same value which means that while the `IsClosed` flag is set to false, in reality it might as well be considered `true`.

**Question:**

I'd like this library to accurately represent the file as read from disk, so I'm going to leave the `IsClosed` flag to `false`, but I'd still like your app to be able to understand if the spline _looks_ like it's closed.  This could be easily accomplished with an extension method, but I'd like your feedback on what that should look like.  My initial thought is something like this:

``` csharp
public static bool IsSplineClosed(this DxfSpline spline)
{
    return spline.IsClosed || (spline.ControlPoints.First().Point == spline.ControlPoints.Last().Point);
}
```

## Text Location Wrong

Can you give me an example of a `TEXT` entity having the wrong location, with both the correct and incorrect values so I can investigate?

Illicitl, 2022-12-04:
Sorry, actually I'm not sure, because when I read this dxf with this library, I read the closed property is wrong, but when I save it as a new dxf with autocad, and read it again with this library, the closed property is normal, and so is the text in this dxf.

brettfo, 2022-12-04:
I think it's still an interesting case where the flags don't report that the spline is closed, but it reality it is.  I've added an extension method `AppearsClosed()` to `DxfSpline` that does basically what I proposed above; returns `true` if either the flag reports that it is closed, or if the first and last control points are equal.

closed 2022-12-04 by brettfo