ixmilia/dxf #191: How to specify a font for Text?

uzername, 2023-01-17:
I'd like to add monospace text to file. There is a `TextStyleName` property, which is the most relevant for my question, but I am not sure how to use it.

brettfo, 2023-01-17:
Presumably this is for display purposes in AutoCAD?  First, I'll say that AutoCAD is _really_ particular about what values it will accept and what ones it won't, so everything below this point is pure theory because I don't have AutoCAD available to test this with, but as I get time I can try it with LibreCAD.

The `TextStyleName` property on a `TEXT` entity needs to have a corresponding entry in the `Styles` property of `DxfFile`, and in that `DxfStyle` entry you'd specify the font in the `PrimaryFontFileName` property.  E.g., something like this:

``` csharp
// n.b., this is untested, but you get the idea

var file = new DxfFile();

var myTextStyle = new DxfStyle("MY_STYLE_NAME");
myTextStyle.PrimaryFontFileName = "the-font-name-that-i-want-to-use";
file.Styles.Add(myTextStyle);

var text = new DxfText("this is a text entity");
text.TextStyleName = myTextStyle.Name; // you could also directly set it to "MY_STYLE_NAME"
file.Entities.Add(text);
```

uzername, 2023-01-23:
Btw, it's the best way to add text - declare your style and then use it anywhere you need a text in dxf file. The STANDARD style does not explicitly specify a font to use, and sometimes Autocad shows wrong characters with STANDARD style.

tumbelo, 2025-01-03:
@uzername Would it be possible to copy your solution here? I'm struggling with same issue and can´t get monospaced strings. Thanks in advance!

uzername, 2025-01-03:
@tumbelo 
```
var sampleTextStyle = new DxfStyle("stylestyle");
sampleTextStyle.PrimaryFontFileName = "Consolas";
someFileOutput.Styles.Add(sampleTextStyle);

DxfText captText = new DxfText(new DxfPoint(20, 20, 0), 10, "Here is text");
captText.TextStyleName = "stylestyle";
someFileOutput.Entities.Add(captText);
```

where someFileOutput is DxfFile to be saved

uzername, 2025-01-03:
By the way, it's resolved already

closed 2025-01-03 by uzername

brettfo, 2025-01-04:
Thanks for confirming this, @uzername.  I was able to quickly verify that I can write an R14 drawing and specify the font.  I've added [an example](https://github.com/ixmilia/dxf/blob/e07bbdcc07966b0317c5defcca7c50c6b0a9cbea/src/Examples/TestExamples.cs#L73) showing this.