ixmilia/dxf #196: Opening file error "Not a valid DXF file header: `?0`"

Siro2002, 2023-04-06:
While i'm opening a dxf file with enconding code 65001 -> exception
" Not a valid DXF file header: `?0`  "

at IxMilia.Dxf.DxfTextReader.GetCodePair() in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfTextReader.cs:line 158
   at IxMilia.Dxf.DxfTextReader.<GetCodePairs>d__5.MoveNext() in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfTextReader.cs:line 27
   at IxMilia.Dxf.DxfBufferReader`1.Advance() in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfCodePairBufferReader.cs:line 33
   at IxMilia.Dxf.DxfCodePairBufferReader..ctor(IDxfCodePairReader reader) in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfCodePairBufferReader.cs:line 48
   at IxMilia.Dxf.DxfFile.LoadFromReader(IDxfCodePairReader reader) in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfFile.cs:line 267
   at IxMilia.Dxf.DxfFile.Load(Stream stream, Encoding defaultEncoding) in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfFile.cs:line 213
   at IxMilia.Dxf.DxfFile.Load(String path, Encoding defaultEncoding) in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfFile.cs:line 184

The same file with encoding 1251 -> no problem, correct visualization

Siro2002, 2023-04-07:
> Presumably you passed in the appropriate encoding to DxfFile.Load()? Can you share the 65001 encoded file with me so I can look into it? You can either upload/attach the file to this issue, or if you want it to remain private, you can email it directly to me at `brett.forsgren@outlook.com` and I'll make sure the file doesn't get out and I'll delete it when I'm done.

I attach 2 files, the same piece exported, but with two different encoding code (the name of the file)
Thank you very much
[Encoding.zip](https://github.com/ixmilia/dxf/files/11177164/Encoding.zip)

brettfo, 2023-04-08:
Thank you for filing the issue and attaching the drawings!  The code that trims off the byte order mark wasn't fully complete so there was a stray `0xFEFF` at the start of the file.  I've updated `main` with the fix.

closed 2023-04-08 by brettfo

Siro2002, 2023-05-22:
Hi,
thank you for your fix, but i'm using ixmilia as a nuget packet in Visual Studio and i see only version 0.8.3. 
Can you make a release? Currently, I can't open dxf files.
Thank you very much!

brettfo, 2023-05-22:
I'd be happy to do another NuGet release, but I want to be sure I've fixed your issue.  Could you try building this repo locally and directly referencing the project/assembly to see if it works?  If you have .NET 7 installed, you can simply run `build-and-test.cmd` (or `build-and-test.sh` if on Linux/Mac) and after that you can add a project reference in your solution.

Siro2002, 2023-05-24:
I tried your code and it seems to work fine. I had to change a line of code in the "DxfFile.cs" file because otherwise it would tell me "file not found". Can you please check? I used the file I gave you last time and Dxffile.load works fine.

my changes have been: 

        public static DxfFile Load(Stream stream, Encoding defaultEncoding)
        {
            int readBytes;
            var firstLine = GetFirstLine(stream, defaultEncoding, out readBytes);
            var reader = new BinaryReader(stream);

            // check for binary sentinels
            DxfFile file;
            if (firstLine == DxbReader.BinarySentinel)
            {
                file = new DxbReader().ReadFile(reader);
            }
            else
            {
//var dxfReader = GetCodePairReader(firstLine, readBytes, reader, defaultEncoding); -> not works
                file = LoadFromReader(new DxfTextReader(reader.BaseStream, defaultEncoding, firstLine));
            }

            return file;
        }

if you think everything is ok, that's fine with me. so i can use nuget package
Thank you very much

Siro2002, 2023-06-16:
> @Siro2002 I'm unclear on your second-to-last comment where I asked you to verify if the current state of `main` worked for your scenario. You responded with:
> 
> > I tried your code and it seems to work fine.  I had to change a line of code...
> 
> Did you have to make any changes to the current state of `main` to get your scenario to work? If not, I'm happy to push an updated package to NuGet, but if any changes are required, I'd need to make those before I hit publish.

This line
        var dxfReader = GetCodePairReader(firstLine, readBytes, reader, defaultEncoding); 
in function
       public static DxfFile Load(Stream stream, Encoding defaultEncoding)

Please check because if I remove that line my scenario will works, otherwise I have a crash in that point. Please check if it is important for other reasons, if not you can make a release.
Thank you

commit 6dd7b1c7 (2023-04-08): fix reading BOM when UTF-8 encoding is explicitly specified

Fixes #196.