ixmilia/dxf #202: DxfEllipse - thickness missing

APOSWE, 2023-12-14:
Hello!
I'm using the IxMilia.Dxf-nuget and i wondered, why the stroke of the ellipse is thicker. It looks like the property thickness is missing in the DxfEllipse?
Then i was looking at the source on github and the whole class is missing here? (src/IxMilia.Dxf/Entities)
Can you help me?

brettfo, 2023-12-14:
You're correct that the `Thickness` property is not on the `DxfEllipse` class, but it sounds like that property isn't what you're looking for.  In a DXF file, an entity's thickness isn't how thick or thin the line is, rather it's more like an elevation, or how tall it is.

If you're looking to modify how thick the entity is drawn on the screen or paper then you likely want to use the `LineweightEnumValue` property.  This value is described under code `370` [here](https://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-3610039E-27D1-4E23-B6D3-7E60B22BB5BD).  The value is a `short` that corresponds to the width of the line in milimeters * 100, e.g., a value of `25` means `0.25mm`.  N.b., the `LineweightEnumValue` property is only honored on DXF `R2000` or newer files, so if you're using this library to create a file you'll have to do something like this:

``` csharp
DxfFile dxf = new DxfFile();
dxf.Header.Version = DxfAcadVersion.R2000;
// ...
DxfEllipse ellipse = new DxfEllipse(...);
ellipse.LineweightEnumValue = 75; // 0.75mm
// ...
dxf.Entities.Add(ellipse);
// ...
dxf.Save("path/to/file.dxf");
```

APOSWE, 2023-12-15:
ok thank you!
and why is the source on github not complete?

brettfo, 2023-12-15:
What do you mean by not compete? The [spec for ellipse](https://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-107CB04F-AD4D-4D2F-8EC9-AC90888063AB) doesn't list thickness as an available property.

On a related note, code is never complete and always able to be improved and this library is something I do in my spare time so it doesn't get 40 hours/week of work.

APOSWE, 2023-12-15:
Sorry, I worded myself a bit unclearly! You're doing a great job! And if I can help in any way, please let me know!

I tried to find out exactly the difference in order to accurately describe the issue!
There I was looking for "DxfEllipse" under the path which i saw in the VS.
[image]

And in github i couldn't find that class "DxfEllipse"... that was why I was asking if the source is not complete.
[image]
That's what I meant by not source complete...
I hope I have been able to clear up the misunderstandings a little 😬

brettfo, 2023-12-15:
Ah, I see.  Much of the code is generated at build time; look at the `IxMilia.Dxf.Generator` project.  The properties for `DxfEllipse` are defined [here](https://github.com/ixmilia/dxf/blob/98b1ed65564e4a05a259efcfc5beabe83b7699d3/src/IxMilia.Dxf.Generator/Specs/EntitiesSpec.xml#L508-L528) and the generator builds the necessary code for the entity.

APOSWE, 2023-12-16:
really great and clever approach! Hopefully I can bother you a few more times if I have a question...  :-)