ixmilia/dxf #204: Support for MLeader

Taylor-Reid, 2024-02-27:
Hi!
 I saw that MLeader is currently not supported by ixmillia, I was curious how difficult it would be to add support for that, and what would be needed? If I knew what was involved in adding support for a new entity type, I'd even be open to trying to add the support myself and add it to your repo, if that is acceptable to you.

brettfo, 2024-02-27:
I'm unlikely to get to `MLEADER` any time soon, but new entity types are added by filling out the relevant section in [`EntitiesSpec.xml`](https://github.com/ixmilia/dxf/blob/645ce4c52f1a3c5221321e22df49ec72a8dfe9e0/src/IxMilia.Dxf.Generator/Specs/EntitiesSpec.xml#L901) and re-running the `build-and-test.cmd` script at the root (because much of the code is generated from that XML file).

`MLEADER` entities appear pretty complex (hence, I haven't done it yet) and it will need many of the methods to be hand-crafted, instead of generated.  The nearest example I can give to a complex entity that had some combination of generated and hand-crafted code was for [`HATCH`](https://github.com/ixmilia/dxf/commit/c658e7c78a244c3ea1956de2f0f0422dfa7ab9ee).

The specification for `MLEADER` entities can be found [here](https://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-72D20B8C-0F5E-4993-BEB7-0FCF94F32BE0&v=2023).

Taylor-Reid, 2024-02-29:
Thank you, I'll take a look.