ixmilia/dxf #206: DxfHatch exception

pgallik, 2024-06-12:
The following error occurred when reading a dxf file with Autocad version AC1021, 

[line that produced the error](https://github.com/ixmilia/dxf/blob/645ce4c52f1a3c5221321e22df49ec72a8dfe9e0/src/IxMilia.Dxf/Entities/DxfHatch.cs#L63)

**The error**
> System.InvalidOperationException: Sequence contains no elements
   at System.Linq.ThrowHelper.ThrowNoElementsException()
   at System.Linq.Enumerable.Last[TSource](IEnumerable`1 source)
   at IxMilia.Dxf.Entities.DxfHatch.TrySetPair(DxfCodePair pair) in D:\a\dxf\dxf\src\IxMilia.Dxf\Entities\DxfHatch.cs:line 30
   at IxMilia.Dxf.Entities.DxfEntity.ApplyCodePair(DxfCodePair pair) in D:\a\dxf\dxf\src\IxMilia.Dxf\Entities\DxfEntity.cs:line 285
   at IxMilia.Dxf.Entities.DxfEntity.PopulateFromBuffer(DxfCodePairBufferReader buffer) in D:\a\dxf\dxf\src\IxMilia.Dxf\Entities\DxfEntity.cs:line 324
   at IxMilia.Dxf.Entities.DxfEntity.FromBuffer(DxfCodePairBufferReader buffer) in D:\a\dxf\dxf\src\IxMilia.Dxf\Generated\Entities\DxfEntityGenerated.cs:line 506
   at IxMilia.Dxf.Blocks.DxfBlock.FromBuffer(DxfCodePairBufferReader buffer) in D:\a\dxf\dxf\src\IxMilia.Dxf\Blocks\DxfBlock.cs:line 213
   at IxMilia.Dxf.Sections.DxfBlocksSection.BlocksSectionFromBuffer(DxfCodePairBufferReader buffer) in D:\a\dxf\dxf\src\IxMilia.Dxf\Sections\DxfBlocksSection.cs:line 79
   at IxMilia.Dxf.Sections.DxfSection.FromBuffer(DxfCodePairBufferReader buffer, DxfFile parentFile, DxfAcadVersion version) in D:\a\dxf\dxf\src\IxMilia.Dxf\Sections\DxfSection.cs:line 60
   at IxMilia.Dxf.DxfFile.LoadFromReader(IDxfCodePairReader reader) in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfFile.cs:line 274
   at IxMilia.Dxf.DxfFile.Load(Stream stream, Encoding defaultEncoding) in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfFile.cs:line 210
   at IxMilia.Dxf.DxfFile.Load(String path, Encoding defaultEncoding) in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfFile.cs:line 186
   at IxMilia.Dxf.DxfFile.Load(String path) in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfFile.cs:line 179
   at Gisvl.Gecko.Processing.Pipelines.Pipelines.Delivery.Features.Dxf.DxfFeatureCollectionProviderActor.<>c.<HandleAddDataSourcesCommand>b__4_1(String filePath) in C:\Users\patri\source\gecko\Processing\Gisvl.Gecko.Processing.Pipelines\Pipelines\Delivery\Features\Dxf\DxfFeatureCollectionProviderActor.cs:line 42

pgallik, 2024-06-13:
Apparently a fix has been made:

https://github.com/ixmilia/dxf/commit/0f578a0642788617321424c65350c6d0bbdef6f0

Is it possible to make a minor release for this to be available? Its not a small bug.

brettfo, 2024-06-13:
@pgallik Version 0.8.4 has been pushed to NuGet as of commit 645ce4c52f1a3c5221321e22df49ec72a8dfe9e0.  NuGet is still validating the package, but it should be live in a few minutes.

closed 2024-06-13 by brettfo