ixmilia/dxf #209: IxMilia.Dxf assembly in NuGet package is unsigned

NicolasJPosey, 2024-12-11:
I am trying to use the IxMilia.Dxf NuGet package in an application that requires signed assemblies. When I try to call the Load method, I get an error saying that the IxMilia.Dxf assembly or one of it's dependencies could not be loaded and that a strongly-named assembly is required. The Public Key Token is null for the IxMilia.Dxf assembly so I am suspecting that it is not signed.

If this is the case, are there any plans to release a signed IxMilia.Dxf assembly in the future?

Thanks in advance.

brettfo, 2024-12-11:
There are no plans to sign the assembly in the package, but if that's something you require I'd recommend building a private copy and signing with your own key.

If you go this route, you'll need to use the `build-and-test.cmd` script (optionally with the `-notest` argument) because there is a pre-build step where code is generated, meaning you can't just call `dotnet build` from a fresh clone and have it work.

You'll also need the 8.0 version of the .NET SDK to do the actual build.

closed 2024-12-11 by brettfo

NicolasJPosey, 2024-12-11:
Thanks for getting back so quickly! I will give your suggestions a try and see where it leads.